sleemanj/JQ6500_Serial #1: getStatus() always returns STOPPED

buttim, 2015-11-30:
I'm using the FullDemo example with an Arduino Uno talking to the JQ6500 via a 3/5V level converter.
Whenever i give the help (?) command the status of the JQ6500 is read and reported on the serial console. But even while playing it says "stopped".
The JQ6500 module I'm using has no microSD card, just internal memory containing a single MP3

sleemanj, 2015-11-30:
It is known that the status command is unreliable  (because the device seems not very reliable when reporting it's status), although I don't recall seeing it always returning stopped (the opposite in fact), see the documentation comment for getStatus() here:

   https://github.com/sleemanj/JQ6500_Serial/blob/master/JQ6500_Serial.h#L260-L271

You can see the implementation of getStatus() here:

  https://github.com/sleemanj/JQ6500_Serial/blob/master/JQ6500_Serial.cpp#L126-L143

You could try commenting out Line 136 in JQ6500_Serial.cpp where it says
   
    if(stat == 0) return 0; // STOP is fairly reliable

in case your device's STOP is not as reliable as the ones I tested on.

You could also tweak the `MP3_STATUS_CHECKS_IN_AGREEMENT` value in `JQ6500_Serial.h`

  https://github.com/sleemanj/JQ6500_Serial/blob/master/JQ6500_Serial.h#L67-L76

to see if that helps.

sleemanj, 2015-11-30:
Actually don't just comment out that line 136 instead you'd want to move it underneath the for loop to line 139, otherwise the while would be an infinite loop.

buttim, 2015-11-30:
Thanks for you (quick!) reply. I'll have a try later and let you know

buttim, 2015-11-30:
Apparently nothing changed moving line 136 to 139. Seems like it is hanging sometimes

sleemanj, 2015-12-01:
You could enable the debug statements to see what is being returned from the device,

  https://github.com/sleemanj/JQ6500_Serial/blob/master/JQ6500_Serial.h#L78

set to 1 to enable debug statement output, you'd be looking for lines in the debug output which are I think starting with `7E 0 42 FF`

(42 is the status command)

sleemanj, 2015-12-01:
Also it's possible that 50ms 

  https://github.com/sleemanj/JQ6500_Serial/blob/master/JQ6500_Serial.cpp#L281

is not long enough for the response, you could try increasing it.

buttim, 2015-12-01:
At 150ms it works! I tried 100 but was not enough.
I now see lots of timing notifications which did not appear before.
I have some problem with volume greater than approx 15, even if I'm using a PSU to power the module at 3.3V. The playback stops when it reaches a loud point in the track. What did you use to power the module?

sleemanj, 2015-12-01:
The power supply to Vcc should be between 3.5 to 5v, I use 5v , the datasheet indicates optimal is 4.2v.  I expect a lower voltage supply will give issues like that. 

I imagine that the JQ6500 has a regulator built in, so a too-low voltage may be below it's drop-out.

Note however that the RX pin is not entirely 5v tolerant, either keep the level shifter on that pin, or just use a 1k resistor in series on it.

See this:
  https://github.com/sleemanj/JQ6500_Serial#connecting-to-your-arduino

NikolaiRadke, 2016-01-22:
Dear James,

thanks for your great work.
I tried to run the module through Vcc with severel voltages, but i encountered  too much different problems. Now i power the device through USB with an upcycled USB plug and 5V - It works perfect.