sleemanj/JQ6500_Serial #5: Random play issue

goldenshuttle, 2016-11-29:
Hi Sleeman. Must thank you a lot for the JQ6500 info. it worked great after my horrible experience with shitty WTV020 and shittier WT588D...
regarding the Random play code you gave, worked great, but can you please tell me which line in the code adjusts the time gap between playing each file ?

sleemanj, 2016-12-02:
You could add a `delay( somethinghere );` on Line 80

https://github.com/sleemanj/JQ6500_Serial/blob/master/examples/RandomPlay/RandomPlay.ino#L80

closed 2016-12-02 by sleemanj

goldenshuttle, 2016-12-06:
Thanks Sleeman, indeed ur reply is very useful. I cannot thank you enough.

S.J.
Electronic Signature Matrix

https://github.com/sleemanj/JQ6500_Serial/blob/master/examples/RandomPlay/RandomPlay.ino#L80

-

goldenshuttle, 2016-12-06:
I do not use Delay() bcoz it wil halt other loops & functions I am adding to the code...So if I use millis() instead of Delay()...I should place it same at line 80 is it ?

S.J.
Electronic Signature Matrix

https://github.com/sleemanj/JQ6500_Serial/blob/master/examples/RandomPlay/RandomPlay.ino#L80

-

sleemanj, 2016-12-06:
I can't answer that without seeing how you intend to implement your delay.