sleemanj/JQ6500_Serial #6: Solution for missing upload software

NikolaiRadke, 2018-03-16:
Dear James,

at first, thanks for your great work!

Most of the 32MBit modules (i reckon every module) brought from ebay come without the upload sortware. Me and several other users tried out your software with the modified config.ini, but it's still not working (Windows 10). It seems like the software has to be on-chip.

Another user wrote a great Linux tool to upload MP3 files and even flash a new (patched) MusicDownload.iso onto the module. It's working great!

Since there are a lot of users without any Linux experience, i decided to create a small (20 MB) bootable ISO-Image (Both UEFI and BIOS) to "repair" the module with just one push of a button. Maybe you find it useful.

https://github.com/NikolaiRadke/JQ6500-rescue-tool

Best wishes from Berlin,

Nikolai

sleemanj, 2018-03-16:
Hi and thanks for that I will add a link to your repo somewhere so people can find it :-)

Note that the JQ6500-16p comes in 8, 16, 32, 64 and 128 MBit versions, of course get a compatible memory chip and I suppose you can drop in as much as you want within reason :-)

The only difference in the software appears to be the size in the ini file.

Attached is a Zip file which I found on a youtube comment from somebody which includes various things including a rar with ISOs for each of those sizes.

[JQ6500 voice chip module.zip](https://github.com/sleemanj/JQ6500_Serial/files/1821040/JQ6500.voice.chip.module.zip)

sleemanj, 2018-03-17:
FWIW, the 64MBit ones I have on my desk have red leds, as do the 16MBit ones, I dont' think LED colour will be a totally reliable way to tell them apart.

It's more reliable (if needing a magnifying glass) to just look at the markings on the SPI flash IC, it's the larger (wider/taller) of the two on the underside.

At least on the ones I have....

The 64MBit flash is KH 25L6406E
The 16Mbit flash is KH 25L1606E

I have a 28P handy which has a KH 32L3206E on it.

I'd assume that the other sizes are similar, KH 25x8... and KH 25x128...

Here is a page which describes the marking of these ICs...
https://www.elnec.com/en/device/Macronix/KH25L6406E+%5BSOP8%5D/

NikolaiRadke, 2018-03-17:
Thanks for your advices!

I wasn't able to get the ISOs from the attached Zip running, but i managed to patch a working 16MBit-ISO for all other devices via hex-editor. 

As you suggested, it was quite easy, i searched for the size string "val=0x200000" and changed the sizes for the depending modules. 128MBit was a bit tricky, i had to delete another byte from a comment a few bytes above, the ISO has to be exactly the size 256KB!

Further more, i changed the volume labels in the ISO and the internal Joilet in order not to confuse the users.

At last, i changed my writings about flash size. You are absolutley right, the LED color has nothing to to with the size. I found a 16 MBit with a blue LED - as part of an 32MBit order... **sigh**. So i changed the text according to your link.

My regarts to the other side of the world.

koogar, 2019-09-16:
> Hi and thanks for that I will add a link to your repo somewhere so people can find it :-)
>
> Note that the JQ6500-16p comes in 8, 16, 32, 64 and 128 MBit versions, of course get a compatible memory [...]

I found the same window flash files as you :) 

I altered the ISO to include the english version of the uploader in all the versions from 8 to 128.

Also included is a cheat sheet, which I made for the for the JQ-6500_28P module

[](https://drive.google.com/open?id=1Kt1FV0__WelMJlX1oZE-hFp9kx6NCMFq)

Thanks for all your hard work!!!