sleemanj/JQ6500_Serial #8: Real data is different than the protocol document

roadcontrolsystems, 2019-01-04:
Hi,
When I debug the data which is sent by the arduino, I see that message frame is the short version of the frame indicated in the datasheet. Version and the checksum fields are missing.

What should I do to run the module with full frame communication?

Thanks.

sleemanj, 2019-01-04:
Is the device you have not functioning with the command frames the library sends?

roadcontrolsystems, 2019-01-04:
I am sorry by opening this issue. I mixed the the protocol with the MP3-TF-16P MP3 Module which is very similar and extended version of this protocol.

For example, to reset the module, I was expecting to see " 7E FF 06 0C 00 00 00 FE EF EF" but library sends "7E 02 0C 7E ".

closed 2019-01-04 by roadcontrolsystems