sleemanj/JQ6500_Serial #11: missing tool to erase memory of jq6500 16P

hhouben, 2019-09-03:
I m looking for a tool to completely erase the internal memory as it seems that the uploaded files have been scrambled.  When I start a "play file by number" just scrambled snippets from the original mp3 files are randomly played. 
Is there a buffer- or memory overflow of some kind or what may be the cause?

NikolaiRadke, 2019-09-04:
Take a look here: 
https://chiselapp.com/user/rmax/repository/jq6500/index

This is a Linux command line tool. If you don't have a Linux system, take a look here:
https://github.com/NikolaiRadke/JQ6500-rescue-tool

It is the same tool embedded in a ready-to-boot-iso for all systems.

sleemanj, 2019-09-08:
I find when the files go wonky it usually means I need to re-encode them.  Anyway no way it can be a problem with the Arduino library, the library just tells the module to play a file, has no interaction with the file itself.

closed 2019-09-08 by sleemanj