sleemanj/JQ6500_Serial #17: soundfile format

derrgis, 2019-10-02:
Hello,
Sorry if this is not the right place to relate.
I get something good uploading one single 10sec. mp3 soundfile with a 45-85Kbps compression.
However, if I upload more than one file the sound is saturated for each.
This is not a power issue.
What file encoding do someone uses ?
Is there something special to know with file upload.

sleemanj, 2019-10-02:
MP3 Format
The devices can play most Stereo MP3 files.   

I found that a “Joint-stereo” file did not appear to work, but simply re-encoding to an ordinary stereo file was fine – I use the LAME Audio Encoder to re-encode it, “lame -m s inFile.mp3 outFile.mp3”, see below for other tips about re-encoding files.

The following sampling rates are supported:  8 / 11.025 / 12 / 16 / 22.05 / 24/32 / 44.1 / 48 kHz

SD Card Files
The 28P can support (supposedly, not tested) up to 32g Micro SD Card (aka TransFlash / TF), files can be split into up to 100 folders, with 1000 files. 

Any old file names work fine.

It’s unclear which files the 5 “hotkeys” correspond to but I suspect it is the first 5 files in the FAT table (Google for a FAT sorter utility suitable for your operating system), although there is a mention in the datasheet of naming folders 00 .. 99 and the files within the folders 001.mp3 .. 999.mp3 so that may have something to do with it (or might just be convenient to name them like that and use a FAT sorter). 

Tips for Re-encoding files
For the 16P particularly due to the limited space in the on board flash, you may wish (need) to re-encode your MP3 files to shrink them, using a lower bitrate.  The utility program “lame” can do this on the command line easily for Linux and Mac (obviously once you install it) the phone preset reduces the files to 16kHz and 16kbps which can reduce files significantly (with commensurate loss of quality).

$ lame --preset phone inputFile.mp3 outputFile.mp3
Or to really get it down, add -B 8 --resample 8 to the command in order to bring it down to 8kHz and 8kbps.

On Windows, try winLAME

For the 28P if you are using an SD Card, you might as well stick with higher quality and get a bigger SD Card :-)

closed 2019-10-02 by sleemanj

thijstriemstra, 2020-07-15:
> The 28P can support (supposedly, not tested) up to 32g Micro SD Card (aka TransFlash / TF), 

I just tested with a 32GB FAT formatted SD-card, and it's not playing any of the files. I tried 16GB before and that worked fine. I'll more closely examine the differences between the 2 sd-cards.