sleemanj/JQ6500_Serial #19: audio clic at 2000 ms

derrgis, 2019-10-23:
Hello,
Sorry to come here again.
I'm using a mqtt broker to trig the JQ's via an ESP01 (ESP8266 mc)
Everythin'g is now fine except I get an horrible audio crack from the JQ for any sound around 2000 ms after the play command is sent to.
I cant understand if the problem is from my code, or from the mqtt broker, since the blue light on the microcontroller lights up two times : first it trig the sound, then it lights again around 2000 ms and the audio artefact occurs.
I've opened an issue here on knolleary's pubsubclient repos (https://github.com/knolleary/pubsubclient/issues/671)[https://github.com/knolleary/pubsubclient/issues/671]

Here some relevant part of my JQ sketch 
in setup
`
  mp3.begin(9600, 0, 2);
  mp3.reset();
  //mp3.setvolume(volume);
  mp3.setEqualizer(MP3_EQ_NORMAL);

  mp3.setLoopMode(MP3_LOOP_ONE_STOP); // Default, plays track and stops
  mp3.pause(); // en pause au démarrage

// we select the built in source NOT SD card source
  mp3.setSource(MP3_SRC_BUILTIN);
  numFiles = mp3.countFiles(MP3_SRC_BUILTIN);// nombre de sons stockés dans le JQ
  mediaType = MP3_SRC_BUILTIN;// pas de carte SD`

in loop
`//-------------- play JQ6500 audiofile ----------------
 if (global_enabled) {
  mp3.setVolume(volume);
       //  ** NOTE: Checking for STOPPED doesn't work with the builtin memory 
  //      because in that case the devie appears to only return PAUSED, not STOPPED  
  byte stat = mp3.getStatus();
  if(stat != MP3_STATUS_PLAYING) 
  {//soundfile index
    mp3.playFileByIndexNumber(soundfile);
     //play it !
    mp3.play();
  }
 }
 global_enabled = false;
}`

sleemanj, 2019-10-23:
I suggest you enable debug and see if you are sending command at 2s mark.

https://github.com/sleemanj/JQ6500_Serial/blob/master/src/JQ6500_Serial.h#L78

closed 2019-10-23 by sleemanj

derrgis, 2019-10-26:
Wow ! thx ! I learn a lot here.