sleemanj/JQ6500_Serial #21: Questions about decoding debug codes

derrgis, 2019-10-26:
JQ6500 debug tool is really interesting.

However I can't understand why I get some of those information in the monitor.
They are not in my code.
So I wonder if they take place in a chronological order ? Or how they occur
Because indeed, something here causes an audio clic in the speaker.

My loop() is really simple: (relevant code)
`
if (global_enabled) {
  mp3.setVolume(volume);
  byte stat = mp3.getStatus();
  if(stat != MP3_STATUS_PLAYING) 
  {
    mp3.playFileByIndexNumber(soundfile);
    mp3.play();
  }
 }
 global_enabled = false;
}`

But I get lot of information when that code is called :
7E 3 6 19  EF ==> [] 
7E 2 42   EF ==> [] 
7E 4 3 0 1 EF ==> [] 
7E 2 d   EF ==> [] 

line 1
3 = MP3_CMD_PLAY_IDX .... plays index ?? code that do taht comes later for me
6 = MP3_CMD_VOL_SET .... setVolume() OK
19 = ......  What is this, ..... not listed ? ??

line 2
2 = MP3_CMD_PREV ..... Why previous track ??
42 = MP3_CMD_STATUS ...... mp3.getStatus() OK

line 3
4 = MP3_CMD_VOL_UP ... why ?
3 = MP3_CMD_PLAY_IDX mp3.playFileByIndexNumber() OK
0 = ......  What is this, ..... not listed ? ??
1 = MP3_CMD_NEXT ??

line 4
2 = MP3_CMD_PREV ..... Why previous track again ??
d = MP3_CMD_PLAY   mp3.play() OK

sleemanj, 2019-10-26:
read the code of the library

you probably do not need mp3.play

each line in debug is single command with arguments, not multiple commands

James Sleeman

---- On Sun, 27 Oct 2019 00:41:20 +1300 notifications@github.com wrote ----

JQ6500 debug tool is really interesting.

However I can't understand why I get some of those information in the monitor.
They are not in my code.
So I wonder if they take place in a chronological order ? Or how they occur
Because indeed, something here causes an audio clic in the speaker.

My loop() is really simple: (relevant code)
if (global_enabled) { mp3.setVolume(volume); byte stat = mp3.getStatus(); if(stat != MP3_STATUS_PLAYING) { mp3.playFileByIndexNumber(soundfile); mp3.play(); } } global_enabled = false; }

But I get lot of information when that code is called :
7E 3 6 19 EF ==> []
7E 2 42 EF ==> []
7E 4 3 0 1 EF ==> []
7E 2 d EF ==> []

line 1
3 = MP3_CMD_PLAY_IDX .... plays index ?? code that do taht comes later for me
6 = MP3_CMD_VOL_SET .... setVolume() OK
19 = ......  What is this, ..... not listed ? ??

line 2
2 = MP3_CMD_PREV ..... Why previous track ??
42 = MP3_CMD_STATUS ...... mp3.getStatus() OK

line 3
4 = MP3_CMD_VOL_UP ... why ?
3 = MP3_CMD_PLAY_IDX mp3.playFileByIndexNumber() OK
0 = ......  What is this, ..... not listed ? ??
1 = MP3_CMD_NEXT ??

line 4
2 = MP3_CMD_PREV ..... Why previous track again ??
d = MP3_CMD_PLAY mp3.play() OK

sleemanj, 2019-10-26:
More specifically, as noted in source
      // Command structure
// [7E][number bytes following including command and terminator][command byte][?arg1][?arg2][EF]
Eg your line 1, command is 6, argument is 19; line 2 command is 42 no argument; line 3 command is 3 arguments are 0 1; line 4 command is d no argument - which matches exactly what you wrote, set volume, get status, play index, play.  You do not need play, play index already plays as far as I remember

James Sleeman

---- On Sun, 27 Oct 2019 00:41:20 +1300 notifications@github.com wrote ----

JQ6500 debug tool is really interesting.

However I can't understand why I get some of those information in the monitor.
They are not in my code.
So I wonder if they take place in a chronological order ? Or how they occur
Because indeed, something here causes an audio clic in the speaker.

My loop() is really simple: (relevant code)
if (global_enabled) { mp3.setVolume(volume); byte stat = mp3.getStatus(); if(stat != MP3_STATUS_PLAYING) { mp3.playFileByIndexNumber(soundfile); mp3.play(); } } global_enabled = false; }

But I get lot of information when that code is called :
7E 3 6 19 EF ==> []
7E 2 42 EF ==> []
7E 4 3 0 1 EF ==> []
7E 2 d EF ==> []

line 1
3 = MP3_CMD_PLAY_IDX .... plays index ?? code that do taht comes later for me
6 = MP3_CMD_VOL_SET .... setVolume() OK
19 = ......  What is this, ..... not listed ? ??

line 2
2 = MP3_CMD_PREV ..... Why previous track ??
42 = MP3_CMD_STATUS ...... mp3.getStatus() OK

line 3
4 = MP3_CMD_VOL_UP ... why ?
3 = MP3_CMD_PLAY_IDX mp3.playFileByIndexNumber() OK
0 = ......  What is this, ..... not listed ? ??
1 = MP3_CMD_NEXT ??

line 4
2 = MP3_CMD_PREV ..... Why previous track again ??
d = MP3_CMD_PLAY mp3.play() OK

derrgis, 2019-10-28:
Yes, you were right, the audio click was coming from that additionnal play() command  in my code (line 4).  Okay for the commands structure, I'm just starting as said before so now I understand reading comments & source code are important. Thanks fo that pedagogy. You're lib is great to learn how things work.

closed 2019-10-28 by derrgis