sleemanj/JQ6500_Serial #37: JQ6500_Error

1502adi, 2022-01-05:
When I compile code on Arduino Nano I get following error:
no matching function for call to JQ6500_Serial::JQ6500_Serial(int, int)

Please help me to solve this error!

sleemanj, 2022-01-05:
The interface changed, you should instantiate like this now (for SoftwareSerial)/...    SoftwareSerial mySerial(8, 9);
    JQ6500_Serial mp3(mySerial);
 ---- On Wed, 05 Jan 2022 20:10:55 +1300  ***@***.***  wrote ----
When I compile code on Arduino Nano I get following error:
no matching function for call to JQ6500_Serial::JQ6500_Serial(int, int)
Please help me to solve this error!

1502adi, 2022-01-05:
Now it shows another error:
'class JQ6500_Serial' has no member named 'begin'

Thank You !

sleemanj, 2022-01-05:
See the HelloWorld example for how to use it.

closed 2022-01-05 by sleemanj

1502adi, 2022-01-05:
Awesome, thanks!