sleemanj/JQ6500_Serial #42: Looping the single audio

Sameer-hesta, 2023-02-21:
Hello,
I am very new to this, and I have a very basic question.

Is there any way to play a single audio in loop in JQ6500 audio module without Ardrino.
If yes let me know the way.

If no, please let know If I apply single audio looping with Ardrino and upload the the code, can the JQ6500 work wine with looping stand alone or without Ardrino?

sleemanj, 2023-02-22:
See playback modes for manual keyinghttps://sparks.gogo.co.nz/jq6500/index.html#manMessage ID: ***@***.***>

sleemanj, 2023-02-22:
As detailed at the link, set to whole file multi play with appropriate resistor. ---- On Wed, 22 Feb 2023 19:14:50 +1300  ***@***.***  wrote ----
I have already tried that,
Either you just connect K1 with ground for once or let the connection remain for a long time, It only play the audio ones.