sleemanj/JQ6500_Serial #43: JQ6500 Set default volume

beikimajid, 2024-06-04:
Hello, how can I set jq6500 in standalone mode so that its volume is the highest level by default (even after power cut) (volume 30)?

sleemanj, 2024-06-04:
I don't think that is possible.

closed 2024-06-04 by sleemanj

beikimajid, 2024-06-04:
Good. If not, is there a way to design a button for the jq6500-16p that starts playback by pressing it and increases the volume by holding the same button for a few seconds?

sleemanj, 2024-06-04:
No.  You should use a microcontroller to control it over the serial protocol then you can do whatever you want.

beikimajid, 2024-06-04:
If you read [this document](https://sparks.gogo.co.nz/jq6500/index.html), you will notice that there is another comment in the "Simple Control with Manual Buttons To Ground" section

sleemanj, 2024-06-04:
On the 16 you need to use ADKEY method.

beikimajid, 2024-06-04:
> On the 16 you need to use ADKEY method.

on 28 work with microswitch?

sleemanj, 2024-06-04:
All the information is in that document, which I wrote.  There is nothing more to add than is in that document.